OpenTrafficCam/OTCamera #12: How to attach framerate info to .h246 video files?

martinbaerwolff, 2020-10-02:
h246 video files contain resolution info, but they don´t contain framerate info.

How can we make sure that info on framerate is available when converting it to other file formats later in the pipeline?

My ideas:

- [ ] Write framerate to filename when recording (thats what I did so far, using e.g. "FR20" in the file names)

- [ ] Write a text file with same name like the .h246 file, which contains framerate (and other relevant stuff)

frunika, 2020-10-14:
The questions also arises with all the other meta data:

- date
- time
- camera (for calibration purpose)
- project?
- coordinates?

Some of the information are more applicable for the whole camera system and not every single video. But maybe it's easier so safe the meta data per video instead of dealing with different meta data files?

martinbaerwolff, 2020-10-27:
I would also prefer a single metafile. This would also allow changes in the camera system over time to be taken into account directly

martinbaerwolff, 2020-10-29:
We could also think about adding the current calibration parameters as meta data to every recording - then no central database on calibrations would be required.

frunika, 2021-01-29:
`JJJJ-MM-DD_HH-mm-ss_hostname_FRxy.h264`

- xy = framerate

closed 2021-01-29 by frunika