OpenTrafficCam/OTCamera #41: Setup Wifi AP

frunika, 2021-01-13:
- [ ] setting up the Wifi AP itself
- [ ] functions to start and stop

randy-seng, 2021-11-18:
# Turn WiFi On/Off

### Periodically turn on WiFi with a [crontab](https://www.linuxwiki.de/crontab) file
- What is [cron](https://www.linuxwiki.de/cron)
- quick and simple editor for cron schedule expressions (https://crontab.guru)

### Use [rfkill](http://manpages.ubuntu.com/manpages/bionic/man8/rfkill.8.html) to turn on/off the WiFi connection:

#### Ensure having rfkill installed:
```bash
sudo apt update sudo apt full-upgrade
sudo apt install rfkill
```

#### Turn off WiFi:
```bash
rfkill block wifi
```
#### Turn on WiFi:
```bash
rfkill unblock wifi
```
**NOTE:** 
- Not sure if sudo is needed to run the commands
- Might need to [connect to WiFi network](https://askubuntu.com/questions/138472/how-do-i-connect-to-a-wpa-wifi-network-using-the-command-line)

### Via Python

#### [wireless](https://github.com/joshvillbrandt/wireless)
- Currently not maintained
- Last release: Jun 28, 2019

#### [pywificontrol](https://github.com/emlid/pywificontrol)
- Last release:  Oct 24, 2016

#### Use [subprocess](https://docs.python.org/3/library/subprocess.html)

```python
import subprocess

result = subprocess.run(["rfkill", "block", "wifi"])
```
- Not sure if it works when running the command with sudo, as it might require credentials
- hardcoding password -> poses security risk  (https://stackoverflow.com/questions/13045593/using-sudo-with-python-script)

randy-seng, 2021-11-18:
# Set up Wifi Access Point

## [RaspAP](https://github.com/RaspAP/raspap-webgui)
- [Instructions](https://docs.raspap.com/ap-sta/) for the Raspberry Pi Zero W
-  [RaspAP Documentation](https://docs.raspap.com)
- comes with GUI
- Description: "RaspAP is feature-rich wireless router software that just works on many popular Debian-based devices, including the Raspberry Pi. Our popular Quick installer creates a known-good default configuration in minutes on all current Raspberry Pis with onboard wireless."

# How to Make a Captive Portal WiFi Access Point
- This [instruction](https://www.maketecheasier.com/turn-raspberry-pi-captive-portal-wi%E2%80%90fi-access-point/) uses [NoDogSplash project](https://github.com/nodogsplash/nodogsplash)

## [Nodogsplash](https://github.com/nodogsplash/nodogsplash)
### Description
Nodogsplash is a Captive Portal that offers a simple way to provide restricted access to the Internet by showing a splash page to the user before Internet access is granted.
- Documentation [here](https://nodogsplashdocs.readthedocs.io/en/stable/)
- Nodogsplash contains minimal version
- [OpenNDS](https://github.com/openNDS/openNDS) is a fork from **Nodogsplash** and provides a "more sophisticated authentication system"

frunika, 2022-04-30:
Tried to configure RaspAP properly for a few hours. Seems to be a really nice software but I couldn't manage it to activate AP-STA mode.

Finally configured hostapd, dhcpcd and dnsmasq manually.

closed 2022-05-02 by frunika